rodralez/NaveGo #3: Strange free INS coasting error

kmos, 2016-12-13:
Hi,

Firstly, thank you for you work! It's a very useful toolbox and it's awesome that you deliver it under gpl. 

Secondly, I am trying to simulate the INS without GPS calibration to estimate how much grow the error with the reference data. As you can see in the figure, 
[image]
the down component error grow up too much for a free inertial solution. 

if you want to reproduce it, I used the ins_gps.m function without kalman filter.

Thank you

rodralez, 2016-12-13:
Dear kmos,

First of all, thank you very much for your words of encouragement!

With respect to your comment, what you are experiencing is the normal operation of an INS without any aiding sensor. The time the INS starts diverging depends on the quality of the inertial sensors. This time can be around seconds for a MEMS IMU and around hours for a navigation-grade IMU.

Regards.

kmos, 2016-12-14:
Dear rodralez,

thank you for your quick reply. Sorry my fault, I was wrong, but if it seems strange,  It's exactly what I need for the beginning of my master thesis work: I was comparing the ins error without an aiding sensor and I thought that for a MEMS IMU the performance was better than in five seconds, ten meters of error.

Anyway, it could be a nice (new) feature to compare the same imu with different kalman filter.

closed 2019-01-17 by rodralez