rodralez/NaveGo #4: Questions about Allan Covariance unit

Sarahholly, 2017-08-10:
figure 1 is the outcome of 'navego_allan_example.m',figure 2 is gyro specification in datasheet of stim300.
What are the units of arw and vrw, respectively?Can they be identical with those in datasheet?
Thank u.
[image]
[image]

rodralez, 2017-08-10:
Hi Sarah,

> What are the units of arw and vrw, respectively?Can they be identical with those in datasheet?

No, units from this IMU datasheet and NaveGo are different.

NaveGo works in SI (International System) units. 

ARW unit is (rad/s/root-Hz). VRW unit is (m/s^2/root-Hz).

Regards.

Sarahholly, 2017-08-11:
Dear rodralez,
ARW in axis x from NaveGo is 0.0079rad/s/root-Hz, now I convert it to unit in datasheet.
 0.0079rad/s/root-Hz = 0.0079rad/root-Hz = 0.0079 * 57.3 degree/sqrt(1/3600hr) = 27.1602degree/root-hour. But it is not same with that in datasheet.
Am I right?
Thank you.
Regards.

rodralez, 2017-08-11:
Dear Sarah, 

You will find in function imu_err_profile.m how to convert from  deg/root-hour to rad/s/root-Hz:

imu_nav.arw = (imu.arw ./ 60) .* D2R;   % deg/root-hour -> rad/s/root-Hz

Please, do the opposite to find deg/root-hour from rad/s/root-Hz.

On the other hand, bear in mind that the only purpose of NaveGo's Allan variance example is to show a practical implementation of the Allan variance. A 2 hrs static dataset is not long enough to do a correct IMU error profile. For more information, see "IEEE Standard Specification Format Guide and Test Procedure for Single-Axis Interferometric Fiber Optic Gyros", Sec. B.4.2. 

Regards.

Sarahholly, 2017-08-15:
Dear rodralez,

Thank you for your patient replying. I'll read the reference you recommended.

Regards.

closed 2019-01-17 by rodralez