rodralez/NaveGo #7: Function for generating ref.mat

paolozoccarato, 2017-09-16:
Dear Rodrigo,
many thanks for this useful toolbox!

Would it be possible to add also the function for generating the ref.mat data or are you using an external tool for generating it?

Many thanks.

Best Regards.
Paolo

rodralez, 2017-09-22:
Dear Paolo,

Thanks for your comments!

Yes, I used an external tool for the NaveGo example which is not open source. Someone commented me that the following tool is an open trajectory generator: http://www.instk.org/. Please, check it.

Regards.

huazai665, 2018-08-18:
can you supply the exteral tool for trajectory generator?

rodralez, 2018-08-19:
El sáb., 18 ago. 2018 a las 8:32, huazai665
(<notifications@github.com>) escribió:

> can you supply the exteral tool for trajectory generator?

Hello,

No, I cannot. The trajectory generator is copyright software.

Regards.
-- 
Rodrigo Gonzalez

huazai665, 2018-08-19:
can you tell me the trajectory generator software name? I buy it

rodralez, 2018-08-20:
El dom., 19 ago. 2018 a las 11:13, huazai665
(<notifications@github.com>) escribió:

> can you tell me the trajectory generator software name? I buy it

I don't know if this tool is still available for buying.

Please, contact me at rodralez [at] frm [dot] utn [dot] edu [dot] ar,
and we can keep talking about this.

Regards.
-- 
Rodrigo Gonzalez

huazai665, 2018-08-21:
can you share us some trajectory data?

closed 2019-01-17 by rodralez