rodralez/NaveGo #8: Questions about the correlation times

Wangshida, 2017-11-03:
Dear Rodrigo,
        Firstly, thank you for your toolbox!
I am a beginner. I want to know how to determine Gyro correlation times and  Acc correlation times for real-world data? Do you have suggested methods and functions?
 
Many thanks. 
Best Regards.
Shida

rodralez, 2017-11-03:
2017-11-03 13:10 GMT-03:00 Wang Shida <notifications@github.com>:

> Dear Rodrigo,
> Firstly, thank you for your toolbox!
> I am a beginner. I want to know how to determine Gyro correlation times and
> Acc correlation times for real-world data? Do you have suggested methods and
> functions?

Dear Wang,

I recommend you to read my paper "Model validation of an open-source
framework for post-processing INS/GNSS systems". There you will find a
brief explanation of the Allan variance and several references. You
can get this paper for free online.

Regards.
-- 
Rodrigo Gonzalez

Wangshida, 2017-11-04:
Dear rodralez,

Thank you. I'll read the reference you recommended.

Regards.

closed 2019-01-17 by rodralez