rodralez/NaveGo #10: Sign of the matrix DCMbn_n in ins_gps.m

ufengger, 2018-03-13:
Dear Rodrigo,

thank you for the useful toolbox!

Maybe my question is very stupid. I am confused about the sign of the matrix _DCMbn_n_ in **ins_gps.m** from the line 307 to 308. In the matlab code **ins_gps.m**, the sign of _DCMbn_n_ is negative, however, in your paper

- (Gonzalez et al., 2015a) R. Gonzalez, J.I. Giribet, and H.D. Patiño. An approach to benchmarking of loosely coupled low-cost navigation systems. Mathematical and Computer Modelling of Dynamical Systems, vol. 21, issue 3, pp. 272-287, 2015.

the sign of _DCMbn_n_ is positive, cf. eq (21e). I am a beginner in this area, so I am not sure whether I missed something or not.

Thanks again and best regards,

Haifeng Wu

rodralez, 2018-03-14:
2018-03-13 5:39 GMT-03:00 ufengger <notifications@github.com>:

> Dear Rodrigo,
>
> thank you for the useful toolbox!
>
> Maybe my question is very stupid. I am confused about the sign of the matrix DCMbn_n in ins_gps.m from the line 307 to 308. In the matlab code [...]

Dear Haifeng,

Thank you very much for your comments. Your question is a valid one!

NaveGo equation (21e) is based on (Groves, 2008) Eq. 12.80.

The sign of the (DCMbn_n * gps.larm) term depends on how you consider
the direction of the lever arm, from GNSS antenna to IMU (negative) or
from IMU to GNSS antenna (positive).

Since NaveGo considers the lever arm from the GNSS antenna to IMU, I
think the sign in Eq. (21e) is wrong. It should be negative.

Nevertheless, I think it is more common to define the lever arm as the
distance from the IMU to GNSS antenna (positive). Therefore, I think I
will change the sign of (DCMbn_n * gps.larm) for a positive sign and
make very clear that the lever arm should be set from IMU to GNSS
antenna.

Please, check if this last approach provides better results in your experiments.

Regards.
-- 
Rodrigo Gonzalez

ufengger, 2018-03-15:
Dear Rodrigo,

> The sign of the (DCMbn_n * gps.larm) term depends on how you consider the direction of the lever arm, from GNSS antenna to IMU (negative) or from IMU to GNSS antenna (positive).
>
> Since NaveGo [...]

many thanks for your nice explanation! 

> Please, check if this last approach provides better results in your experiments.

Yes, it does improve the results!

Best regards,

Haifeng Wu

closed 2018-03-15 by ufengger