rodralez/NaveGo #11: ref.mat, imu1.mat, imu2.mat, gps.mat

awolinski2012, 2018-04-09:
Could you please post ref.mat, imu1.mat, imu2.mat, gps.mat that works with your latest navego_example in matlab.  I tried to use your code on Octave and FreeMat  however some functions(for example matlabrc, sgolayfilt .)  are not implemented in Octave and FreeMat so I would like to bypass the code which is using them by means of using mentioned ready simulation mat files instead. Thank you.

Sincerely, Adam

rodralez, 2018-04-09:
2018-04-08 23:25 GMT-03:00 awolinski2012 <notifications@github.com>:

> Could you please post ref.mat, imu1.mat, imu2.mat, gps.mat that works with your latest navego_example in matlab. I tried to use your code on Octave and FreeMat however some functions(for example [...]

Dear Adam,

Yes, NaveGo uses some few functions from MATLAB which are not
supported by Octave yet.

I have added imu1.mat, imu2.mat, and gps.mat to NaveGo's repository.

Regards.

awolinski2012, 2018-04-09:
Dear Rodrigo,

Thank you for reply and  postings in repository.

Sincerely, Adam

closed 2019-01-17 by rodralez