rodralez/NaveGo #13: Definition of body frame XYZ?

xelmirage, 2018-07-31:
Hi~I'm working with your wonderful work to process the data from my own UAV autopilot. But Now I am confused with the relation of the Body frame XYZ and the roll/pitch/yaw of the UAV.  I tried to found the general relationship and there are many possibilities existing. So I'm wondering if there is any further definition of the relation of the Body frame XYZ and the roll/pitch/yaw of the UAV?

rodralez, 2018-08-01:
> Hi~I'm working with your wonderful work to process the data from my own
> UAV autopilot. But Now I am confused with the relation of the Body frame
> XYZ and the roll/pitch/yaw of the UAV. I tried to [...]
Hello,

Body frame in Navego is positive X-forward, Y-right, and Z-down.

Regards.
--
Rodrigo Gonzalez

xelmirage, 2018-08-05:
Thank you for your explanation. There is another issue. The navego_example.m seems to have some mixing use of radian and degree in the IMU error profile section. So I am a little confused about the actual unit of every parameter. Now I am guessing the correct units. if you don't mind, later I may create a pull request to fix this problem...

rodralez, 2018-08-06:
2018-08-04 23:45 GMT-03:00 xelmirage <notifications@github.com>:

> Thank you for your explanation. There is another issue. The navego_example.m seems to have some mixing use of radian and degree in the IMU error profile section. So I am a little confused about the [...]

Hello,

NaveGo works internally using SI units. So, angles must be represented
in radians and turn rates in radians per second.

Then, you can choose to show the attitude errors between NaveGo and a
reference (true) dataset in radians or degree. I choose degrees
because is a more familiar unit for me.

I strongly recommend you to read my two articles cited in the
Publications section at NaveGo repository:

(Gonzalez et al., 2015) R. Gonzalez, J.I. Giribet, and H.D. Patiño.
NaveGo: a simulation framework for low-cost integrated navigation
systems, Journal of Control Engineering and Applied Informatics, vol.
17, issue 2, pp. 110-120, 2015.

(Gonzalez et al., 2015a) R. Gonzalez, J.I. Giribet, and H.D. Patiño.
An approach to benchmarking of loosely coupled low-cost navigation
systems. Mathematical and Computer Modelling of Dynamical Systems,
vol. 21, issue 3, pp. 272-287, 2015.

Regards.
-- 
Rodrigo Gonzalez

xelmirage, 2018-08-06:
Now I see，I had some misunderstanding of the code, thank you for your explanation.

closed 2019-01-17 by rodralez