rodralez/NaveGo #14: Can not find "a_std" in acc_gen when run navego_allan_example.m

FoxLins, 2018-08-15:
Hi Rodralez,

When I tried to run navego_allan_exmaple.m. I have got following errors:

Reference to non-existent field 'a_std'.

Error in acc_gen (line 101)
    a_wn(:, i) = imu.a_std(i).* wn(:,i);

Error in navego_allan_example (line 159)
fb = acc_gen (ref, ustrain);   % Generate acc in the body frame

Could you please check this issue for me whether there is some problem in stim300.mat file?

Thank you very much!
Fox

rodralez, 2018-08-15:
2018-08-14 23:47 GMT-03:00 FoxLins <notifications@github.com>:

> Hi Rodralez,
>
> When I tried to run navego_allan_exmaple.m. I have got following errors:
>
> Reference to non-existent field 'a_std'.
>
> Error in acc_gen (line 101)
> a_wn(:, i) = imu.a_std(i).* wn(:,i); [...]

Dear Fox,

Thank you very much for pointing out this problem.

I have already solve it. Please, check if everything is OK.

Regards.

-- 
Rodrigo Gonzalez

FoxLins, 2018-08-16:
Hi Rodralez,

I have checked the new file, it's OK now. Thank you very much for your quick fix.

Thank you.
Fox

closed 2019-01-17 by rodralez