rodralez/NaveGo #16: Lever arm for velocities

brumann, 2018-09-03:
I am not sure to understand how you are using the lever arm for the computation of the position.
In fact, I have the impression that you use it for the position but not for the velocity

"    zp = Tpr * ([lat_e(i); lon_e(i); h_e(i);] - [gps.lat(j); gps.lon(j); gps.h(j);]) ...
        + (DCMbn * gps.larm);
    
    zv = (vel_e(i,:) - gps.vel(j,:))';    "  (from ins_gps.m)

Am I wright or did I misunderstand the script ?

rodralez, 2018-09-10:
El lun., 3 sept. 2018 a las 8:10, brumann (<notifications@github.com>) wrote:

> I am not sure to understand how you are using the lever arm for the computation of the position.
> In fact, I have the impression that you use it for the position but not for the velocity
>
> " zp = Tpr [...]

Hi brumann,

You are correct. NaveGo does not correct lever arm for velocity,
mostly because I didn't see any impact when using noisy MEMS inertial
sensors. Maybe it could be a good idea to implement lever-arm velocity
correction for higher grade sensors.

Regards.
-- 
Rodrigo Gonzalez

brumann, 2018-09-11:
Hi Rodrigo,

Thank you very much for the answer, since we are using a high grade IMU it improved a bit the accuracy. 

Regards

closed 2018-09-11 by brumann