rodralez/NaveGo #17: Strange accuracy problem whereas RTK received

brumann, 2018-09-19:
Dear Rodrigo, 

First of all thank you for your code, it is very helpful. 

However I noticed a strange phenomenon, at the beginning of my simulation the hybridization is actually worse than GPS alone. For almost 3min there is about 35cm of error for the hybride trajectory whereas GPS error is below 2cm. Imu data are generated from the reference trajectory, with error estimated in static with Allan-variance. 

What I do not understand is that since GPS.stdm is of about 1cm for this period, the kalman filter should give more weight to GNSS, and it appears not to be the case ? I have spent days on it, I don't see what I am missing. 
I tried to take only gnss when RTK is received, with no success (kalman filter diverge).
Screenshots are attached. 

Thank you in advance for your answer
Sincerely,

[image]

[image]
(The vehicle starts moving after 15sec)

LZNUDT, 2019-01-12:
in my opinion ,the outcome of the integration of INS/GNSS  is worse than RTK at the begining! so you may check your INS algrithm and Kalman Fusion Equation!

rodralez, 2019-01-14:
Dear brumann,

I think you should tune the values of Q and R matrices. Errors in KF tend
to be very sensitive to the values of these matrices.

Regards.
-- 
Rodrigo Gonzalez

closed 2020-04-14 by rodralez