rodralez/NaveGo #20: In consistent tau length

xelmirage, 2019-07-04:
Hi, I'm working with an actual IMU. While generating the error parameters using allan_imu.m, I got inconsistent tau lengths.
From line.176 of allan_imu.m, the for loop repeats 3 times, in the first time, allan_overlap() returns a tau that has 48 items, but in the second time, the allan_overlap() returns a tau that has 48 items. Then the program retuens an error says
`the dimention of left side is 48×1，the dimention of left side is 47×1。 

error allan_imu (line 184)
    imu.fb_tau  (:,i) = tau';`

And I debugged the program, it seems that in line.301 of allan_overlap.m the  program returns 48 and 47 in first and second time.

So is there a walkaround of this problem?

Glad to have your reply:)

rodralez, 2019-07-04:
Hi xelmirage,

The allan_imu() function creates the tau_v vector which defines the
points where the Allan variance will be evaluated by allan_overlap().
This vector is the same for the six inertial sensors mounted on a
particular IMU. So, the tau vector returned by the allan_overlap()
function should be the same for every sensor.

Can you share the IMU data so I can check this strange behavior?

Kind regards.
-- 
Rodrigo Gonzalez

xelmirage, 2019-07-05:
Happy to have your timely reply, the data can be found at:
https://github.com/xelmirage/my_allan

Because github has a limit of file size, I packed it into multi-volume zip, you may need to download all the files and unpack it.

LZNUDT, 2020-03-22:
> 
> 
> Happy to have your timely reply, the data can be found at:
> https://github.com/xelmirage/my_allan
> 
> Because github has a limit of file size, I packed it into multi-volume zip, you may need to download all the files and unpack it.

Hi,friend! 
I have analyzed your data, you can look at the results！
`
For X-axis Gyroscope, the result as follows:
random walks

N =

     1.681055538327384e-04

rate random walks

K =

     2.602532081636726e-07

Bias Instability

B =

     7.689027351904626e-06

BI, correlation time, in seconds\n

ans =

     1.155520000000000e+03
`if you have any question ,please contact me with wechat:13121105069

LZNUDT, 2020-03-23:
you can find the code of allan 
[https://blog.csdn.net/wuwuku123/article/details/104973773](url)

Barak28, 2020-06-18:
Hi,
Please note that _vel_update_ corrects the specific force by subtracting g as a vector (3*1) and not by a scalar as written in function description (moreover,  _fn_c_ dimensions are (3*1)..so g must be (3*1) ) 
Thanks!

rodralez, 2021-04-03:
> Happy to have your timely reply, the data can be found at:
> https://github.com/xelmirage/my_allan
> 
> Because github has a limit of file size, I packed it into multi-volume zip, you may need to download all the files and unpack it.

Hello,

I am very sorry for the late reply.

I processed your static data with the last version of NaveGo. Everything seems to be fine. Next, the Allan variance plots:

[image]
[image]

Kind regards.

closed 2021-04-05 by rodralez