rodralez/NaveGo #21: How to calculate gb_psd and ab_psd

LZNUDT, 2020-03-23:
Hi Dear Rodralez:
I have read most of your articles, and I have carefully studied the previous version of NaveGo code. At the same time, I have also carefully looked at the latest version of the code in recent days. You have done a good job, very Thank you; but now there is a question;
In the paper 《Performance Assessment of an Ultra Low-Cost Inertial Measurement Unit for Ground Vehicle Navigation》Table 1 and Table 2, shows the noise parameters calculated by the allan method!

[image]

In the program 'navego_example_real_ekinox.m',we can see the struct of 'ekinox_imu' , but I don't understand how you calculated ‘gb_psd’ and 'ab_psd' .
From your simulation pragram, i know the 'imu_si.gb_psd = imu_si.gb_dyn .* sqrt(imu.gb_corr)'
but when i  use the dynamic error and correlation time in Table 2 to calculate it ,but i don get the same result with you !
Please help me ,thank!

LZNUDT, 2020-03-23:
Take the calculation of gb_psd as an example：
[image]
What I calculated is not very consistent with yours。
[image]

rodralez, 2020-03-24:
Dear  Zhen,

First of all, thank you very much for your kind words. And thank for
using NaveGo and your detailed examination of our research.

About your question, I checked my code and this is the formula I used
to calculate xb_psd values:

    imu_ekinox.gb_psd = ekinox_allan.gb_dyn .*
sqrt(ekinox_allan.gb_corr) / 2;  % rad/s/root-Hz;
    imu_ekinox.ab_psd = ekinox_allan.ab_dyn .*
sqrt(ekinox_allan.ab_corr) / 2;  % m/s^2/root-Hz;

I divide by 2 the original xb_psd values in order to get better
results when running the Kalman filter. This is the reason.

You can try using original values and divided-by-2 values and compare
the performances of both INS/GNSS systems.

Regards.

LZNUDT, 2020-03-29:
thank you very much!
I look forward to the following areas:
1. Zero speed correction and vehicle model;
2. Tight combination of IMU / GNSS
3. Now that you have mentioned that the IMU gyroscope does not conform to the normal distribution, maybe other types of filters are more suitable for integrated navigation.

rodralez, 2020-04-08:
Hi Zhen,

If you would like to become an active contributor to NaveGo, please send me
an email to the email addresses that is provided in the main page of this
repository.

Kind regards.

closed 2020-04-14 by rodralez