rodralez/NaveGo #27: About the "eps"

kakusang2020, 2020-08-28:
Would you mind telling me about how to calculate the 
eps(time interval to compare current IMU time to current GNSS time vector (s).)

Usually, how can I get it?

rodralez, 2020-08-29:
El vie., 28 ago. 2020 a las 2:44, (<notifications@github.com>) escribió:

> Would you mind telling me about how to calculate the
> eps(time interval to compare current IMU time to current GNSS time vector (s).)
>
> Usually, how can I get it?

Hello kakusang2020,

A rule of thumb for choosing eps is,

gnss.eps = mean ( diff ( imu.t ) ) / 2;

Kind regards.

-- 
Rodrigo Gonzalez

kakusang2020, 2020-09-02:
> El vie., 28 ago. 2020 a las 2:44, (<notifications@github.com>) escribió:
> Would you mind telling me about how to calculate the eps(time interval to compare current IMU time to current GNSS time [...]

closed 2020-09-02 by kakusang2020

kakusang2020, 2020-09-02:
Thanks for your reply! Sorry, I have another question, about the " LOAD REF DATA", I need to use Inertial Explorer, so can you tell me about how to use my imu raw data convert to .imr data?