rodralez/NaveGo #29: Size of g and use in vel_update.m

floosche2, 2020-10-01:
Dear rodralez, 

thank you for your work, I really appreciate your code.

I am a little bit confused since the function "gravity"  creates a vektor g_n with the size [1x3]

```
g = (g0 ./ (1 + (h ./ Ro)).^2);

Z = zeros(size(lat));

g_n = [Z Z g];
```

First of all, in the description of the function its mentioned as size [Mx1]
But more important for me: in the function vel_update you use 

`fn_c = fn - coriolis - g`

fn is size [3x1], coriolis is size [3x1] but g is size [1x3] which actually creates a matrix [3x3] for fn_c (which confuses me since fn_c should have size [3x1] as well).

Thank you very much

floosche2, 2020-10-01:
I got it, you put gn' in the function input! 
Sorry

closed 2020-10-01 by floosche2

rodralez, 2020-10-05:
Dear floosche2, thank you very much for using NaveGo. Please, just let me know if you have any additional question. 

Kind regards.