rodralez/NaveGo #30: size of DCM

AliRamshini, 2020-11-02:
Hi my friend.
I just started learning but I had a problem ...
DCMnb is 3*3 matrix but in ( gyro_gen_delta.m ) code you told DCMnb is N*9 matrix , why?
I don't understand following code:
dcmnb = reshape(DCMnb(k,:), 3, 3); 
dcmnb_old = reshape(DCMnb(k-1,:), 3, 3);

please help me to understand it.  thank you

rodralez, 2020-11-02:
Dear AliRamshini,

Thank you for using NaveGo and for your feedback!

DCMnb is a 3x3 matrix. The function gyro_gen_delta() needs several
DCMnb matrices in order to calculate the delta angles of three axial
gyroscopes.

One way to pass several DCMnb matrices to gyro_gen_delta() is by using
a new matrix where each DCMnb is allocated in one row. Let's call this
new matrix DCMnb_m.

So, each row of DCMnb_m contains the 9 elements of a particular DCMnb
matrix ordered as [a11 a21 a31 a12 a22 a32 a13 a23 a33].

I hope the above is useful to you. Please let me know if you have any questions.

Best regards.

AliRamshini, 2020-11-03:
Thank you for helping me improve...

closed 2020-11-03 by rodralez