rodralez/NaveGo #31: equation

AliRamshini, 2020-11-15:
Hi Dr,
I think one section of your program is wrong , am I true ? 
you programed following equation in line 52 at gnss_err_profile.m :
gnss.std(2) = gnss.stdm(2) / (RN + h) / cos (lat); 

but as you told in your article it must be :
gnss.std(2) = gnss.stdm(2) / ((RN + h) * cos (lat)); 

and I can't find the reference of this equation in the book.
thank you.

rodralez, 2020-11-16:
Dear Ali,

Thank you very much for inspecting NaveGo code so thoroughly.

The mathematical expressions that you point out are equivalent.

Please, try at a MATLAB the following code:

>> a = rand(1), b = rand(1), c = rand(1)
>> d = a / b / c
>> e = a / (b * c)
>> d == e

You will see that d and e are equal.

Best regards.

Rodrigo Gonzalez

AliRamshini, 2020-11-16:
Thank you for your response alot...

closed 2020-11-17 by rodralez