rodralez/NaveGo #37: Error with example

InsJet, 2020-12-17:
Hello! At first I would like to thank you for the great tool!

But I have some trouble when I;m trying to start an example (synthetic-data, Matlab 2014):

NaveGo: generating GNSS synthetic data... 
NaveGo: generating IMU1 ACCR synthetic data... 
Reference to non-existent field 'DCMnb_m'.

Error in acc_gen (line 64)
    acc_b = acc_nav2body(acc_ned, ref.DCMnb_m);

Error in navego_example_synth (line 271)
    fb = acc_gen (ref, imu1);   % Generation of acc in the body frame

rodralez, 2020-12-17:
Hello InsJet,

Thank you very much for your kind words.

It is a very strange error indeed. I have checked the ref.mat file and everything seems to be fine.

Please, download again the ref.mat and check if the DCMnb_m file is present. You should see the following fields:

[image]

Kind regards.

InsJet, 2020-12-18:
Hello rodralez, 
Thank for your answer!And one more question :) Which version of Matlab do you use?

rodralez, 2020-12-18:
Currently I am using MATLAB R2019a, but I think NaveGo should work fine with MATLAB R2014x.

Kind regards.

closed 2020-12-30 by rodralez