rodralez/NaveGo #38: Coordinate systems (vehicle frame)

ravindraji4, 2020-12-30:
Hello Dr. Gonzalez, 

Thanks for the open sourced software. It is stepping stone and great insight for the beginners. 

It is mentioned in the comment section of each example file 
"% NOTE: NaveGo assumes that IMU is aligned with respect to body-frame as
% X-forward, Y-right, and Z-down." 

1. In "navego_example_synth.m" the **Vehicle frame** has same direction as of IMU?
2. For "navego_example_real_ekinox.m" & "navego_example_real_mpu6000.m". This is similar as **1**, except we have lever-arm and we compensate for the lever-arm effect?
3. Does it matter if vehicle move along a different axis. For example, assuming the vehicle is moving along x-axis in case of "navego_example_real_ekinox.m". **In second case**, the vehicle collects data moving along y-axis. Considering such scenario should we change anything? // or we keep our model the same?

Sorry for asking so many questions. 
I wish you a very happy and healthy new year ahead in advance.

Best regards,

rodralez, 2020-12-30:
> Hello Dr. Gonzalez,
>
> Thanks for the open sourced software. It is stepping stone and great
> insight for the beginners.
>
> It is mentioned in the comment section of each example file
> "% NOTE: NaveGo assumes that IMU is aligned with respect to body-frame as
> % X-forward, Y-right, and Z-down."
>
>    1. In "navego_example_synth.m" the *Vehicle frame* has same direction
>    as of IMU?
>
> Hello ravindraji4,

The short answer is no. Each IMU may define what are the positive
directions for accelerations and turn rates for its body frame. Therefore,
you should match NaveGo body-frame sign convention to your body-frame IMU
sign convention.

>
>    1. For "navego_example_real_ekinox.m" &
>    "navego_example_real_mpu6000.m". This is similar as *1*, except we
>    have lever-arm and we compensate for the lever-arm effect?
>
> Yes, real examples should be compensated for the lever-arm effect since an
IMU and a GNSS antenna cannot be placed in the same point in space. In the
synthetic data example it is assumed that an IMU and a GNSS antenna are in
the same point in space for the sake of simplicity, but this situation is
impossible in a real-life scenario.

>
>    1. Does it matter if vehicle move along a different axis. For example,
>    assuming the vehicle is moving along x-axis in case of
>    "navego_example_real_ekinox.m". *In second case*, the vehicle collects
>    data moving along y-axis. Considering such scenario should we change
>    anything? // or we keep our model the same?
>
> I am not sure if I get your point, but I can say that nothing has to be
changed. The body-frame sign convention is independent of the trajectory of
a particular vehicle. Once it is set, the body-frame sign convention is
constant during the operation of the IMU.

Best regards.

Rodrigo Gonzalez

PS: if you have further questions, please, open a discussion at NaveGo
discussion section. Issues section is for reporting bugs in NaveGo's code.

ravindraji4, 2021-01-04:
Thanks for your comments. I will open a discussion in the forum.

closed 2021-01-04 by ravindraji4