rodralez/NaveGo #39: ZUPT algorithm

AliRamshini, 2020-12-30:
Hi , Dr Gonzalez
In first thank you for your best tools in navigation .
I studied your codes but I didn't understand ZUPT algorithms.
please help me to understand it .
Thank you for your attention ...

rodralez, 2020-12-30:
El mié, 30 dic 2020 a las 15:47, AliRamshini
(<notifications@github.com>) escribió:

> Hi , Dr Gonzalez
> In first thank you for your best tools in navigation .
> I studied your codes but I didn't understand ZUPT algorithms.
> please help me to understand it .
> Thank you for your attention ...

Hello AliRamshini,

NaveGo's ZUPT algorithm tries to detect if a vehicle is stopped, i.e.,
if their velocities are under a threshold. If so, current attitude is
equal to the mean of previous attitudes defined by the ZUPT window
time, and current position is equal to the mean of previous positions
defined by the ZUPT window time.

Best regards.

Rodrigo Gonzalez

PS: if you have further questions, please, open a discussion at NaveGo
discussion section. Issues section is for reporting bugs in NaveGo's
code.

rodralez, 2021-04-01:
This was a question about NaveGo's ZUPT algorithm.

closed 2021-04-01 by rodralez