rodralez/NaveGo #46: Kalman filter

yangming19940101, 2021-01-07:
I have read your paper,Nave GO : a simulation framework for low -cost integrated navigation systems. I have a question? formulation:(23.a) have matrix G and matrix U, but In your matlab simulation code,I don't see G and U.  In Kf_prediction.m 
kf.xi = kf.A *kf.xp.  so I am very confused about this question? please help me understand it. hank you for your attention ...

rodralez, 2021-01-08:
> I have read your paper,Nave GO : a simulation framework for low -cost integrated navigation systems. I have a question? formulation:(23.a) have matrix G and matrix U, but In your matlab simulation [...]

Hello, matrices G and F (not U) must be updated before the execution of the Kalman filter. These two matrices are updated in the F_update() function. 

Kind regards.

4nDrEi95, 2021-01-14:
@rodralez F_update() function is missing from the package.

rodralez, 2021-01-16:
> @rodralez F_update() function is missing from the package.

Thank you for pointing out this error. This problem was already solved.

closed 2021-01-16 by rodralez