rodralez/NaveGo #49: 5x Median Absolute Deviation (MAD) criteria

211938554, 2021-01-28:
When I used Allen variance for IMU calibration, if the angular velocity (WB) changed very little, I tested the condition of all 0, so an error would be reported, and the reason was found to be function [retVal, S, errorb, tau] = allan_overlap(data,tau,name,verbose)
% Screen for outliers using 5x Median Absolute Deviation (MAD) criteria
MAD = median (abs (medianfreq) / 0.6745);
if verbose &gt;= 1 && any(abs(medianfreq) &gt; 5*MAD)
Odl = (abs (medianfreq) & gt;5*MAD);
Outliers = data. Freq (odl);
Fprintf (1, 'allan_overlap: OUTLIERS: There appear to be %d OUTLIERS in the frequency data.\n', length(OUTLIERS));      
Fit_line = polyval (s.l inear, (1 / data. Rate: 1 / data. Rate: length (data. The freq)/data. Rate) ') - s.m edian;   
Idl = (medianfreq &lt;(5*MAD + fit_line) );
Data. The freq = data. The freq (idl);
Medianfreq = medianfreq (idl);
 
Fit_line = polyval (s.l inear, (1 / data. Rate: 1 / data. Rate: length (data. The freq)/data. Rate) ') - s.m edian;
Idl = (medianFreq >;(-5*MAD + fit_line) );     &&这行代码删除掉了我全部的数值0
Data. The freq = data. The freq (idl);
Medianfreq = medianfreq (idl);
S.o utliers = length (outliers);
The else
S.o utliers = 0;
end
It will make data filtering, delete all data.freq values, and get empty [], and report an error.

211938554, 2021-01-28:
> 当我使用艾伦方差进行IMU校准时， 如果角速度 （WB） 变化很小，我测试了所有 0 的条件，因此将报告一个错误，并且发现原因是函数 [retVal， S， errorb， tau] = allan_overlap （数据， tau， 名称，
> 详细） % 屏幕使用 5x 中位数绝对偏差 （MAD）
> 标准 MAD = 中位数 （abs （中值） / 0.6745）;
> 如果详细 >= 1 &&和任何 （abs （中位）> 5_ MAD） Odl = （abs （中位） & gt;5_MAD）;
> 异常值 = 数据。弗雷克（欧德尔）;
> Fprintf （1，'allan_overlap：异常值：频率数据中似乎有 %d 异常值。
> Fit_line = 多值 （s.l inear， （1 / 数据。速率： 1 / 数据。速率：长度（数据）。freq）/数据。速率） '） - s.m电;
> Idl = （中位频率 <（5*MAD = fit_line） ）;
> 数据。freq = 数据。freq （idl）;
> 中位数频率 = 中位数频率（idl）;
> 
> Fit_line = 多值 （s.l inear， （1 / 数据.速率： 1 / 数据。速率：长度（数据）。freq）/数据。速率） '） - s.m电;
> Idl = （中位freq >;(-5*MAD = fit_line） ）;0 数据
> 。freq = 数据。freq （idl）;
> 中位数频率 = 中位数频率（idl）;
> S.o uliers = 长度（异常值）;
> 其他
> S. o utliers = 0;
> 结束
> 它将进行数据筛选，删除所有 data.freq 值，并获取空 +，并报告错误。

换句话说，就是这个异常数据筛选，采用这种方法，对于 中位数为0的时候，是否存在不妥。

rodralez, 2021-04-03:
Hello,

Thank you very much for pointing out this problem.

Errors arised when MAD was equal to zero in function allan_overlap(). This issue was solved doing:

<code>
if MAD ~= 0.0<br>
    hm=plot([fx(1) fx(2)],[5*MAD 5*MAD],'-r');<br>
    plot([fx(1) fx(2)],[-5*MAD -5*MAD],'-r');<br>
end<br>
<\code>

Best regards.

closed 2021-04-05 by rodralez