rodralez/NaveGo #70: F_update with infinity imu.gb_corr or imu.ab_corr

YangRiverJiang, 2021-09-19:
Hi Authors,

I was reading the code of NaveGo and I am curious about something. When you update F without valid imu.gb_corr or imu.ab_corr, the matrix of Fbg or Fba is not produced, see lines 170 to 172 and lines 178 to 180 of function F_update. However, when MAG is switched to 'OFF', these two matrices are needed to produce G, see lines 186 to 202. So I wish to know if these were intentionally written? Could you please let me know a suggestion on these two matrices for my processing?
Thank you very much!

River

YangRiverJiang, 2021-09-22:
Hi, I am glad that I am able to offer tiny help to the community, but I just meant that the variable Fbg should be initialized following line 169:     Fgg = Z; which should be similar to lines 171 to 173. So when isinf(imu.gb_corr) returns true, Fbg can still be initialized and used to create matrix G.
The same comment on the variable Fba, which is just next.
Thank you very much,

River

rodralez, 2021-09-23:
Hi River, 

Now I understand your point. I was probably trying different formulas for Fbg and Fba. Now these matrices are equal to the identity matrix.

Please, check the last commit and let me now if everything is OK.

Regards.

closed 2021-09-23 by YangRiverJiang