rodralez/NaveGo #82: calibration and non-orthogonality

scott198510, 2022-01-04:
add calibration processing for IMU data ?  And also add frame transformation  due to level arm effect？

rodralez, 2022-01-17:
Hello,

Thank you very much for suggesting these new features.

It could be nice to a add laboratory calibration procedure.  

On the other hand, frame transformation due to level arm effect is already implemented. See lines 330 and 334 at function ins_gnss.m (branch develop). Or maybe you are suggesting something else.

Kind regards.

SaeedSafarian, 2024-01-03:
Hello,
I have a question about calibration, it is not obvious for me how to obtain ab_psd/gb_psd from datasheet of two module ADIS16400/ADIS16405.
These two parameter also known as acceleration/gyroscope random walk.