rodralez/NaveGo #107: Checking potencial problem in function F_update.m, relates to #101

rodralez, 2022-03-06:
F_update.m
    The state transition matrix in the program is 15 by 15, which is different from the original paper,this in itself has nothing to say.

In the program：
F = [F11 F12 F13 DCMbn Z ;
F21 F22 F23 Z -DCMbn ;
F31 F32 F33 Z Z ;
Z Z Z Fgg Z ;
Z Z Z Z Faa ;
];

G = [DCMbn Z Z Z ;
Z -DCMbn Z Z ;
Z Z Z Z ;
Z Z Fbg Z ;
Z Z Z Fba ;
];

why not:

F = [F11 F12 F13 -DCMbn Z ;
F21 F22 F23 Z DCMbn ;
F31 F32 F33 Z Z ;
Z Z Z Fgg Z ;
Z Z Z Z Faa ;
];

G = [-DCMbn Z Z Z ;
Z DCMbn Z Z ;
Z Z Z Z ;
Z Z Fbg Z ;
Z Z Z Fba ;
];

Only in this way it is reasonable. Is there any special treatment or transformation?
It is also the latter rather than the former in corresponding paper about NaveGo.

rodralez, 2022-03-07:
This is an outdated version of the function. Please, check the last version.

scott198510, 2022-03-07:
@rodralez  The last version is : 

F = [F11 F12 F13 DCMbn Z ;
F21 F22 F23 Z DCMbn ;
F31 F32 F33 Z Z ;
Z Z Z Fgg Z ;
Z Z Z Z Faa ;
];

G = [DCMbn Z Z Z ;
Z DCMbn Z Z ;
Z Z Z Z ;
Z Z Fbg Z ;
Z Z Z Fba ;
];

This is  not right,  no minus sign in the last verison.

The right one is :  

F = [F11 F12 F13 -DCMbn Z ;
F21 F22 F23 Z DCMbn ;
F31 F32 F33 Z Z ;
Z Z Z Fgg Z ;
Z Z Z Z Faa ;
];

G = [-DCMbn Z Z Z ;
Z DCMbn Z Z ;
Z Z Z Z ;
Z Z Fbg Z ;
Z Z Z Fba ;
];

rodralez, 2022-03-07:
I had to add again a negative sing to the 2do DCMbn in matrices F and G because MPU-6000 example stopped working. If you try this example with your proposed matrices F and G, the INS/GNSS diverges badly.

closed 2022-03-07 by rodralez

scott198510, 2022-03-08:
@rodralez  Can you tell me some  explanations?

I have refered to the books,

[image]

rodralez, 2022-03-08:
In this case, Farrellś model shows to be working. You may compare both models, Farrell's and Groves', and thus determine what's going on.

scott198510, 2022-03-09:
@paolozoccarato   If you try this example with your proposed matrices F and G, the INS/GNSS diverges badly.

I have used the model as my proposed matrices, the result is a little different, but it doesn't diverge.

commit fb438652 (2022-03-07): Several bugs are fixed, close #104 close #107
commit b636d024 (2022-03-07): Merge pull request #110 from rodralez/develop

Several bugs are fixed, close #104 close #107