rodralez/NaveGo #116: Add DCM from b-frame to v-frame

scott198510, 2022-03-28:
Usually, an IMU frame is known as the b-frame. It is assumed that the b-frame is aligned with the v-frame, as shown below.

Assuming that the two frames have a roll angle q, pitch angle f, and azimuth angle , the
DCM for these angles can be written as

[image]

scott198510, 2022-03-31:
@rodralez

rodralez, 2022-03-31:
Hello, yes, adding this new DCM matrix could be a very good improvement.

You can create a new branch from `develop`, maybe called `imu-to-body-dcm`, then develop this new imu-to-body DCM, and finally push the new branch to NaveGo repository.

Please, just let me know if you want to become a NaveGo collaborator and I will allow to be part of our project.

Kind regards.