rodralez/NaveGo #117: Canada example for Visual navigation is not working

rodralez, 2022-03-30:
The `navego_example_canada_data.m` has been tested and is not working.

If `FusionCase = "inertial_gnss"` option is selected, the system diverge.

If `FusionCase = "inertial_visual_gnss"`,  the following error arise:

```
Error using horzcat
Dimensions of arrays being concatenated are not consistent.

Error in ins_visual_gnss (line 499)
            kf.z = [zv' zp']';

Error in navego_example_canada_data (line 62)
        nav_e = ins_visual_gnss(imu_planetary,gnss_planetary,visual_planetary,'dcm');
```
Please, download the current branch `visual-inertial-gnss` and work using the files from these folders. Avoid using code located at other folders.

rodralez, 2022-04-27:
Since.txt files were not provided but the .mat did, `navego_example_canada_data.m` script was modified to
run using the .mat files only.  And these *mat data files were upload at folder `NaveGo/examples/planetary-data/canada-planetary-data/data/` in order the Canada visual example keeps running.

johanndiep, 2022-04-27:
Thanks Rodrigo for the input. I realized I forgot to add the `.txt`

rodralez, 2022-04-27:
Johan, yes, please, remove all my changes to the original script if .txt are added.

On the other hand, please check if the error in ins_visual_gnss (line 499) keeps showing.

johanndiep, 2022-04-27:
I assume the error should be gone once the required input files are provided. 

Note: as of currently, I am not able to test it since I do not have access to Matlab anymore. However, with the provided `.txt` files, the results should be similar to the results displayed in the following video: 
[](https://www.youtube.com/watch?v=sPnPMBs1gSU&t=135s)

rodralez, 2022-04-27:
> Note: as of currently, I am not able to test it since I do not have access to Matlab anymore.

If you don't have access to MATLAB, this is a problem. Maybe you can try Octave instead.

commit ca704567 (2022-05-15): several minor errors were fixed, closes #117
closed 2022-05-15 by rodralez