rodralez/NaveGo #120: Quaternion corrections is wrong?

scott198510, 2022-04-05:
% Quaternion corrections
    % Crassidis. Eq. 7.34 and A.174a.
    antm = [0.0 qua(3) -qua(2); -qua(3) 0.0 qua(1); qua(2) -qua(1) 0.0];
According to Crassidis. Eq. 7.34 and A.174a.

antm = skewm((qua(1) qua(2) qua(3))

Then we get:
antm = [0.0 -qua(3) qua(2); qua(3) 0.0 -qua(1); -qua(2) qua(1) 0.0];

There is just a minus sign difference between the code and your paper/book.

@rodralez Doc. , can you give me some explanations?

scott198510, 2022-04-05:
@rodralez

rodralez, 2022-04-07:
Hello, 

You are right. According to Crassidis' equations, `antm` should be positive. Besied, I think I took this portion of code from Crassidis MATLAB examples (http://www.acsu.buffalo.edu/~johnc/second_ed/), instead of transforming his equations to MATLAB code. Besides, if `antm` is set as positive, NaveGo diverges.

commit 017fd297 (2022-04-07): quaternion equations match Crassidis' formulas, fix #120 fix #121
closed 2022-04-16 by rodralez