rodralez/NaveGo #122: MPU6000 NaveGo example is not working

rodralez, 2022-04-07:
After some improvements of NaveGo's code, MPU6000 NaveGo example stopped working correctly. Please, check if signs of acc sensor or gyros sensor have to be changed.

scott198510, 2022-04-08:
I also found this problem，by checking your sensor data/gnss data, I think your accelerometer static bias is wrong,The static bias of acceleration in the vertical direction is about -9.8,

This is VERY wrong. The so-called static bias does not mean the value at the static time, but the measured value under the noise state relative to the ideal measured value, 
Therefore, if the measured value in the ideal state is the local gravitational acceleration value, such as -9.81, then the measured value with noise in the static state is -9.815, then we can get the static bias is -0.015.

When I run my program with - 9.8, I get the wrong result, but I change it to a small value as I mentioned above, everything is fine!

scott198510, 2022-04-08:
> After some improvements of NaveGo's code, MPU6000 NaveGo example stopped working correctly. Please, check if signs of acc sensor or gyros sensor have to be changed.

As explained in your paper, the measurement with wave sign is the measurement with noise, and the measurement without wave sign is the ideal measurement。

The actual measured value and ideal value of acceleration with noise are the same as that of gyroscope

[image]
[image]

rodralez, 2022-04-08:
> I also found this problem，by checking your sensor data/gnss data, I think your accelerometer static bias is wrong,The static bias of acceleration in the vertical direction is about -9.8,
> 
> This is VERY wrong. The so-called static bias does not mean the value at the static time, but the measured value under the noise state relative to the ideal measured value, Therefore, if the measured value in the ideal state is the specific value, such as -9.81, then the measured value with noise in the static state is -9.815, then we can get the static bias is -0.015.
> 
> When I run my program with - 9.8, I get the wrong result, but I change it to a small value as I mentioned above, everything is fine!

You are right! The problem was solved by changing the ACC Z static bias.

Regards.

scott198510, 2022-04-10:
@rodralez  Yes, I found the signs have changed.

for  example:

imu.wb(2,:)' = -0.0789    0.0223    0.1951  % old data( master branch)

imu.wb(2,:) = 0.0789    0.0223    0.1951  % latest data( master branch)

Signs of gyros sensor is different. Maybe the old one is correct.

rodralez, 2022-04-16:
Yes, the sign of Gyro X (mpu6000_imu.wb(:,1)) was changed to improve the performance of the INS/GNSS system.

commit 259f2d66 (2022-04-08): MPU6000 example is fixed, fix #122
closed 2022-04-16 by rodralez