rodralez/NaveGo #145: Question in generating the gyro about g_err_b

chengjunlongwhusgg, 2022-08-16:
g_err_b(i,:) = ( omega_in_b * gyro_b (i,:)' )' or gyro_err_b(i,:) = ( omega_in_b )';

rodralez, 2022-08-16:
`omega_in_b` is the projection of both Earth and transport rates in the body frame. 

It is the inverse operation of: 

`wb = (wb - DCMbn' * (om_ie_n + om_en_n));  % Titterton, Eq. 3.29, p. 32` 

in `att_update()`.

chengjunlongwhusgg, 2022-08-16:
邮件已收到！！！

chengjunlongwhusgg, 2022-08-17:
Outputs from three ideal orthogonal gyroscopes are expressed as:
<p align="center">
<img src="https://latex.codecogs.com/svg.image?\omega&space;_{ib}^{b}=\omega&space;_{nb}^{b}&space;&plus;&space;R_{n}^{b}\left&space;(&space;\omega&space;_{ie}^{n}&plus;\omega&space;_{en}^{n}&space;\right&space;)" title="\omega _{ib}^{b}=\omega _{nb}^{b} + R_{n}^{b}\left ( \omega _{ie}^{n}+\omega _{en}^{n} \right )" />
</p>
omega_in_b is the projection of both Earth and transport rates in the body frame, ie
<p align="center">
<img src="https://latex.codecogs.com/svg.image?omega\_in\_b&space;&space;=&space;R_{n}^{b}\left&space;(&space;\omega&space;_{ie}^{n}&plus;\omega&space;_{en}^{n}&space;\right&space;)" title="omega\_in\_b = R_{n}^{b}\left ( \omega _{ie}^{n}+\omega _{en}^{n} \right )" />
</p>
What does the "g_err_b" means in code?  The finall output is
<p align="center">
wb_sim = gyro_b + g_err_b + g_wn + gb_sta + gb_dyn + g_rrw
</p>
gyro_b is same as omega_in_b? What does the code "g_err_b(i,:) = ( omega_in_b * gyro_b (i,:)' )'" means?

chengjunlongwhusgg, 2022-08-19:
Any idea?

rodralez, 2022-08-22:
Hello, 

I think we've got a bug here. Thank you very much for spot it. 

I modified files `gyro_gen.m` and `acc_gen.m` in branch `develop`.

Please, check them and let me know your comments.

chengjunlong0524, 2022-08-23:
Thanks for your reply. In noise_rrw.m, it probably should be
<p align="center">
rrw_n (j, i) = rrw_n(j-1, i) + rrw(i) * sqrt(dt) .* b_noise(j-1);
</p>
<p>
See that the unit of rrw may be rad/s/root-Hz, rad/s^2/root-Hz, m/s^2/root-Hz or m/s^3/root-Hz. In another form, it is rad/sqrt(s), rad/s/sqrt(s), m/s/sqrt(s) or m/s^2/sqrt(s).
</p>
<p>
In addition,  angle rate random walks(rad/s^2/root-Hz or rad/s/sqrt(s)) seems also reflects bias instability. Are gb_dyn  and gyro_rrw  repeated? This question also applys to acc_gen. 
</p>

closed 2022-10-23 by rodralez

chengjunlong0524, 2022-10-23:
邮件已收到！！！

commit f91a1353 (2022-08-22): fixed bug in gyro simulation, closes #145